mkalten/reacTIVision #2: Fiducial marker angle is inaccurate

simpae, 2015-01-29:
The angle for a fiducial marker is seemingly only updated after a certain minimum angle delta is reached.
The verbose output suggests that reacTIVision is capable of detecting a more accurate angle as the angle values reported are not in fixed intervals. Sometimes after rotating a marker back and forth a few times, the angle values sometimes lie between previous values, suggesting marker recognition is able to determine a more accurate angle.

Is there any way to improve angle accuracy by myself or this an issue with reacTIVion/marker recognition itself?

norpol, 2015-01-29:
How are you testing the angles?

simpae, 2015-01-29:
I am just reading the marker angle from the TUIO message.
The verbose console output by reacTIVision also shows these fairly huge value jumps (around 4deg). The reported angle value does not change for small changes of the rotation of the physical marker.

simpae, 2015-02-01:
Found it: [FiducialObject.cpp](https://github.com/mkalten/reacTIVision/blob/6fdb0349a2b017ae353a86998281a5d73f6ba11f/common/FiducialObject.cpp#L137-139)

I understand this is probably meant to reduce the number of TUIO messages and/or to reduce jittery angle values, but being able to configure this degree theshold would be nice.

mkalten, 2015-02-01:
Hello, this is actually a never fixed workaround to reduce angle jitter, in general libfidtrack provides an angle accuracy below 1 degree. For the next version I am already planning to implement a proper noise filter based on the 1€ filter (http://www.lifl.fr/~casiez/1euro/) for both the angle and position values, which should provide much better accuracy and performance compared to the current simple 1pixel and 3 degree thresholds.

mkalten, 2016-03-11:
OK, this is finally fixed in the most recent developer version of reacTIVision 1.6. This version now generally has a precision of 0.5 pixels for the position and 0.5 degrees for the rotation.

In order to get rid of possible noise, I added a OneEuroFilter option, which still needs to be tested for optimal configuration.

closed 2016-03-11 by mkalten