mkalten/reacTIVision #3: Add support for V4L2_PIX_FMT_GREY

floe, 2015-03-25:
Some Video4Linux drivers (such as the experimental one for the SUR40 :-) directly provide an 8-bit greyscale image with format code V4L2_PIX_FMT_GREY. It's probably sufficient to just add this in https://github.com/mkalten/reacTIVision/blob/master/linux/V4Linux2Camera.cpp#L56 and https://github.com/mkalten/reacTIVision/blob/master/linux/V4Linux2Camera.cpp#L420 but I'm not familiar enough with portvideo to tell if that is indeed enough...

norpol, 2015-03-26:
I would appreciate this also!
If you need someone to test it on the SUR40, we have one lying around here.

mkalten, 2015-03-26:
Sure, I'll take care of it, that part of the code needs some cleanup too apparently ...
Norpol, actually we are looking for a SUR40 donation for these efforts!

mkalten, 2015-03-26:
BTW, for backwards compatibility it also should be quite straightforward to add a MS byte tag tracker, using the existing libfidtrack segmentation data ;)

mkalten, 2015-03-26:
https://msdn.microsoft.com/en-us/library/ee804885%28v=surface.10%29.aspx

norpol, 2015-03-26:
@mkalten I don't think we can donate it, since it was a donation by some stranger of our hackerspace in Vienna. But if you'd like to drop by, just tell me.

mkalten, 2015-03-26:
sure, I'll check it out when I'm in Vienna next time ... but are you actually actively using it for anything else than a coffee table?

norpol, 2015-03-26:
At the moment no one got enough time hacking around with it. :disappointed:

mkalten, 2015-05-07:
OK, I am back ;)
the V4L2_PIX_FMT_GREY pixelformat is now handled by V4Linux2Camera
let me know if this works for you ...

norpol, 2015-05-07:
Super cool! I'm going to test it within a few days, lots of other things to do right now. Thank's a lot!

floe, 2015-05-08:
+1 for the picture :+1: 

The module relies on the input_polldev kernel thread for also fetching video data; that was intentional to keep both streams in sync. Therefore, if you completely disable the input device, the video stream also stops. However, if you just detach it using `xinput float $SUR40`, then the video stream is still usable.

mkalten, 2015-05-08:
Wow, it's great to see that everything is working now! Many thanks for testing, and most of all thanks to Florian for implementing the V4L support on the SUR40!

closed 2015-05-08 by mkalten

floe, 2015-05-08:
BTW @norpol: how many FPS do you get?

norpol, 2015-05-08:
Hi, I can push you the update later today. It's just too bright outside to test anything.

mkalten, 2015-05-15:
OK, I finally also got hands on a SUR40 and already managed to test everything myself. On Ubuntu using a 4.1rc2-realtime kernel I actually get around 46fps ... good job @floe !

Here are a few tips to configure  reactivision to work properly with the pixelsense image:
* background subtraction (empty table, fullscreen F1, spacebar)
* tresholder: gradient (G) and tilesize (cursor down) both at maximum
* finger size (F) at 20 and finger sensitivity (cursor down) at 100
* set the CPU governor to maximum to increase CPU speed for lower latency

mkalten, 2015-05-15:
good news @floe, I now managed to get the full 60fps from pixelsense just by reducing the POLL_INTERVAL to 4ms in sur40.c

floe, 2015-05-15:
*facepalm* @mkalten thanks a lot for figuring that out! Didn't see the wood for the trees, apparently ;-)

mkalten, 2015-05-15:
I just created a pull request for your SUR40 driver, which also implements two further V4L functions for frame size and frame rate query

nylki, 2017-05-30:
> Oh nice it works! Just had to compile the recent kernel module.
@norpol In case you are reading this, could you elaborate? I am trying to get a raspi-cam working with reactivision 1.5.1 as well as 1.6.0 but I always get `no V4Linux2 devices found`.

mkalten, 2017-05-30:
this seems to be the wrong thread, but you'll need to load the according V4L2 modules to access the raspicam via video4linux2: sudo modprobe bcm2835-v4l2

mkalten, 2017-06-21:
@norpol btw, check out the latest version of @floe sur40 kernel driver ...
we recently implemented more V4L2 controls, which now allow to set up the overall image brightness, contrast and gain ... this basically replaces the windows calibration and works much smoother!

norpol, 2017-06-21:
Really nice, will check out!

mkalten, 2017-06-21:
good @norpol , just one more remark:
the new module works with kernels 4.9 and later ..
and you'll need to load it with the option sur40_video=true