mkalten/reacTIVision #5: Support for PS3Eye on Yosemite?

jceipek, 2015-04-03:
Since that camera seems to be recommended for this kind of tracking application, how hard would it be to add support through something like https://github.com/inspirit/PS3EYEDriver given that the macam drivers are deprecated and no longer work?

segabor, 2015-04-23:
Hi,

I started working on enabling PS3Eye on Mac. You can see my progress here https://github.com/segabor/reacTIVision-mkalten/tree/ps3eye
It's working more or less. One thing needs to be fixed, refreshing framebuffer by a separate thread.

Gábor

jceipek, 2015-04-28:
Thanks, @segabor! I just got it working in my project.

segabor, 2015-04-28:
You're welcome, @jceipek !

mkalten, 2015-05-07:
hi segabor, many thanks for your contribution!
I just included most of your code (manually) to the master branch, in general your code works just fine, but with my PS3Eye, I only achieved 30fps for VGA and 60fps for 320x240.

segabor, 2015-05-08:
@mkalten my pleasure!

I suppose ps3eye.cpp does not allow higher framerate. I think, magic byte sequences used to control the device only support a few combinations of framerate and resolution. But that's just an assumption.

closed 2015-05-15 by mkalten