mkalten/reacTIVision #6: WebSocket support

floe, 2015-04-23:
I'm opening this issue as a kind of reminder for myself, since we are planning on implementing WebSocket support in a current project. Pull request hopefully coming soon :-)

mkalten, 2015-05-07:
OK, I already integrated the WebSocket code into the TUIO11 and TUIO20 libraries. I guess it would be great to add some browser examples to test everything. I'd suggest to implement a console TuioDump and graphical TuioDemo example, just like for the other environments. In any case this issue should go to the according TUIO repositories ...

floe, 2015-05-08:
But the Reactivision codebase doesn't actually contain the WebSocket class, right? It's only part of the TUIO reference implementation right now - I was assuming that Reactivision could open a listening WebSocket by default and wait for incoming connections, parallel to the standard UDP transport.

mkalten, 2015-05-08:
Yes, reacTIVision is still cooking its own TUIO messages without using the actual library. But I will start with the integration of the TUIO11 or TUIO20 libraries now, although I am not decided yet which I should use. TUIO11 would allow testing within many existing environments, but TUIO20 would allow the integration and implementation of many new features ....

floe, 2015-05-08:
Ah, now I see. In this case, I would also suggest to switch to TUIO 2 for the next rTV release, and perhaps add a conversion tool?

mkalten, 2015-05-08:
In both cases I'll need to reorganize the current code base substantially, so integrating TUIO11 for an intermediate 1.6 release would allow me the proper testing of everything in a stable environment, while directly jumping to TUIO20 would make that difficult for the time being. The intermediate reorganization will allow a rather quick switch to the TUIO20 code for a following 2.0 release, which may need a longer development phase anyway, since it requires the implementation of several new tracking features.

The reorganization mostly applies to the acquisition layer (portvideo) as well as the transport layer (TUIO) and some other improvements such as a better calibration and the integration of filters.

The new tracking features will concentrate on the possibility of using various fiducial types such as the current amoebas and my new yamarashi symbols as well as the integration of other types such as QR codes or an implementation of the MS byte tags for example. This kind of feature will definitely require TUIO20 already, so a switch will be necessary at this point. Also the additional contour tracking of untagged objects will not only require TUIO20 but also involve the implementation of a better threshold algorithm, which may result to be a tricky task.

From this perspective the intermediate 1.6 release based on TUIO11 still makes sense, since all those intermediate steps are anyway necessary towards the 2.0 release, and should keep things more stable on the way.

floe, 2015-05-10:
:+1: sounds very sensible, as the APIs of TUIO 1.1 and 2.0 are not that different.

RitterLean, 2015-11-13:
I'd like to know what the current status of the development is. I have difficulties with the realization of a multitouch table web application on a OS X main computer connected via wifi and hdmi with a Sur40 with Linux (which is the touch input device and main screen)  and reacTIVision because I don't know yet how to get touch events into my browser (on my OS X). I already tried npTuioClient which was promising but it was not possible for me to get proper html5 touch events for any interaction (touch, drag etc.) For instance like in D3.js. So the next reasonable step would be the usage of websockets?

mkalten, 2016-04-05:
with the integration of the TUIO11 library, the WebSocket support is now also available from within reacTIVision. It can be configured by changing the according type tag in reacTIVision.xml:

<tuio type="web" port="3334">

This still needs the according implementation and testing on the web client side ... anyone?

floe, 2016-04-06:
@nomve has most of the client-side implementation done in https://github.com/mmbuw/gispl/tree/master/source/tuio - should also work standalone AFAICT?

mkalten, 2016-04-06:
OK that's great, I will have a look at it. Should be a good starting point for the TUIO/Web client implementation. Maybe you could give it a try too, checking out the latest reacTIVision 1.6 from master.
Things are wrapping up nicely now, under Linux the most significant improvements are already working.

nomve, 2016-04-06:
Hi,

I can confirm it's generally working, but I haven't used it too much (mostly because of #13). Maybe worth noting is that there can be only one source; I tried just adding `<tuio type="web" port="3334">` to the config, and it didn't work before I deleted the udp entry.

mkalten, 2016-04-06:
Thanks for testing this Goran! Yes, at the moment it is only possible to select one TUIO transport at once, but I will add multiple transport options asap. This is now just a configuration issue, since the TuioServer already support multiple transport alternatives.

mkalten, 2016-04-06:
OK, I added the possibility for multiple TUIO transport configurations with the last commit,
let me know if my comment on #13 solved your camera configuration problem ...

nomve, 2016-04-07:
Thanks again. I'll report any issues if I run into them, but in general the WebSocket connection works fine.

mkalten, 2016-04-07:
OK, then I will close this issue for now ... it would be great if you could prepare a minimal demo based on your code, which includes the handling of TUIO11 cursors, objects and blobs. I'd be glad to include this into the official collection of TUIO11 reference implementations. thanks, Martin

closed 2016-04-07 by mkalten

nomve, 2016-04-11:
>  it would be great if you could prepare a minimal demo based on your code, which includes the handling of TUIO11 cursors, objects and blobs. 

The original Tuio.js library had a couple demos that should still work, but I need to clean them up a bit. But will do.

Jones-S, 2023-11-20:
I am using the release, cause I did not want to dive into how I can build 1.6. If there was an easy guide I can also build 1.6 with xcode...

gmenegoz, 2023-11-20:
I think (but I am just a random guy using this software) that websocket support is not present in 1.5.1.

I usually work on Arch Linux and Windows 10. In both OS I tried 1.5.1 and I compiled 1.6 from source code.
In 1.5.1 only UDP, in 1.6 I found websocket too.

Since I actually needed websocket in 1.5 too, I had to employ a node package to receive UDP traffic and send it to a websocket.

Maybe I'm missing something but this is my experience