mkalten/reacTIVision #7: Linux and Sur40

RitterLean, 2015-11-09:
Hey Martin,

we wanted to use reacTIVision on a Samsung Sur40. The problem is that the touch screen (which is the current camera) ist not found by the application. Multitouch is working on Linux though. We disabled touch for the system enviroment so it was only possible to use mouse and not your actual hands and touches. This step didn't solve our problem. We know that because we used VLC Media Player to capture the grayscale video output of the touch surface when it was enabled and after disabling touch in Terminal it did not record anything at all. Do you have any idea how to enable the camera for reacTIVision?

mkalten, 2015-11-09:
Make sure to use a Linux Kernel from version 4.2 or later (such as included in Ubuntu 15.10), which actually provides the necessary V4L2 driver for the SUR40. Then use the updated reacTIVision from Github, which also works with the grayscale buffer provided by the device driver.

RitterLean, 2015-11-09:
thank you, we are getting grayscale image now. But the multitouch gestures are interfering with the interaction itself. We used Compiz already to reduce possible interaction like opening other applications or tabbing out of reacTIVision it self but is there any possible way to only keep reacTIVision only?

mkalten, 2015-11-09:
you can turn off the multitouch driver with the xinput command:
xinput float #dev
(get the SUR40 device number by just running xinput from the terminal)

RitterLean, 2015-11-12:
thank you it worked

closed 2015-11-12 by RitterLean