mkalten/reacTIVision #10: Android support

victordiaz, 2016-01-12:
It would be awesome to use mobile devices for fast prototyping purposes. How difficult would it be to port it?

mkalten, 2016-01-14:
Hello Victor, thanks for getting back to me on the topic!
Basically you'll need to follow these steps to support a new platform:
* use the devel tree, since it includes the latest camera API code
* portvideo is actually now a separate project, which can be found in ext/portvideo
* create a now android folder,  for all the platform specific code
* create a SportVideo example project (a simple video viewer, see any other platform)
* for android you will need to use the native SDK and use the according SDL2 library for the GUI
* implement a single subclass of a CameraEngine (see the other platforms for the quite simple API)
* make sure your SportVideo project behaves as it should regarding camera configuration etc.
* in case you want to avoid SDL2, you can also implement a subclass of the UserInterface
* when everything works, you can create an equivalent android project for reacTIVision
* although the rest of the code should then compile fine, you may need to add some platform specific #ifdefs

I hope that helps for the start,
maybe we can have a chat on the topic elsewhere ...

simone27zero1, 2017-05-08:
hello to everybody...mkalten i ask you a specific about your comment: all steps you descrive in your post, must i do on my phone or on my pc and after send them on my phone?

mkalten, 2017-05-08:
Hello @simone27zero1, the steps we describe above actually refer to the development process which is necessary to provide Android support through PortVideo in the future. Unfortunately this process has not started yet, therefore neither the current nor the upcoming reacTIVision 1.6 won't be available for Android yet. Nevertheless the plan to port reacTIVision to mobile phone platforms still is interesting, although probably I'll start with the iOS port first, since this seems easier than Android.

simone27zero1, 2017-05-08:
mkalten...then isn't it possible download reactivision for androind?

mkalten, 2017-05-08:
no

simone27zero1, 2017-05-08:
pity that is not there! but i have another question: can i use the camera of mobile phone to read the marker and send the ID of marker to processing with the TUIO library? beacuse i download processing for android and the libraries...

mkalten, 2017-05-08:
If you want to try TUIO on Android, you can use TuioDroid for multi-touch interaction:
https://github.com/TobiasSchwirten/tuiodroid

While reacTIVision is not available for Android, I am also not aware of any other application,
which would provide fiducial marker tracking through TUIO on this platform ...

simone27zero1, 2017-05-08:
thanks a lot! now i download it and try to do that i would like to do... thanks again for the help you gave me

simone27zero1, 2017-05-09:
hello! Always about reactiviosion topic on the mobil phone, if instead of using reactivision, that doesn't exist on the mobil phone, used the camera of phone and through the app TUIODroid send camera data to processing? i hope do it!

mkalten, 2017-05-09:
I am sorry to destroy your imagination, but this doesn't work as you think. I'd suggest that you read a bit into the topic, since it seems that you are a bit on the wrong track here ...

simone27zero1, 2017-05-09:
okok nothing, thanks anyway

sonicdebris, 2019-09-15:
Hello, I hacked together an android app. The current status of my work is at:
https://github.com/sonicdebris/reacTIVision/tree/reactivision-android

The app is very basic, all settings are to be set in the code, there's no UI feedback except for the processed image, but it does manage to send data from the phone to the `TuioDemo` client reference implementation on a PC.

@mkalten if this is interesting for you I could open a pull request once I have cleaned up things a bit and added at least some in-app settings.

RomualdMashup, 2019-09-17:
@sonicdebris Wahoo, this look amazing ! Thank you for sharing.

sonicdebris, 2019-09-17:
@RomualdMashup ahah you're welcome! It's still _very_ rough, there's no calibration and setting the destination IP requires a restart. I plan to make it nicer, in the meanwhile let me know you happen to try it!

morusa, 2020-02-26:
hi @sonicdebris 

I am tested your android app on several devices but it does not correctly identify the symbols.
Is it configured to identify the amoeba set?
How can I change the resolution of the camera?

thanks for sharing your work :)