mkalten/reacTIVision #13: camera.xml

nomve, 2016-04-06:
Hello,

I've downloaded the latest (1.6) version and although it works, I can't change the camera settings in camera.xml like I could before. Any change I make (e.g. camera id, size) gets overwritten once I start reactivision. Any ideas on what the problem there could be?

mkalten, 2016-04-06:
Hi Goran, reacTIVision generally saves all changes that are done in the interactive configuration during runtime back to camera.xml. In case your camera.xml contains any invalid configuration it would fall back to the default settings. Since most settings (including choosing cameras and image formats) can now be done interactively, I'd suggest doing an initial configuration from the running application. Key O allows to configure all image features, key K allows to select one of the available cameras and image formats.

nomve, 2016-04-07:
Hi Martin,

thanks, selecting the camera with K works perfect.

closed 2016-04-07 by nomve