mkalten/reacTIVision #14: Calibration Grid

jominga, 2016-04-18:
Hello,

I used the grid calibration option of Reactivision because I'm using a Fisheye camera and I also want to ignore some parts of the picture captured by it. However the positions of the blobs I receive in my client are the same before and after the calibration.

Am I missing an intermediate step?
Thanks in advance!

[image]

mkalten, 2016-04-18:
Hi, the current development branch still doesn't correct the distortion for finger and blobs. Since this is quite trivial to add, I implement the according updates as soon as possible ...

mkalten, 2016-04-18:
I just pushed a quick update including the according min_size configuration option for the blobs,
let me know if this works for you ... I didn't have time to test it much yet

jominga, 2016-04-19:
It works! Thank you very much!

mkalten, 2016-04-25:
@jominga ... finally I managed to commit the proper blob/finger calibration update as well.
let me know if everything works as expected, so we can close this issue

jominga, 2016-04-26:
It works like charm. Thank you very much! 

And I will move the camera closer, thanks for the advice!

closed 2016-04-26 by mkalten