mkalten/reacTIVision #15: Grid result deforming

PabloArty, 2016-04-19:
Hello MK! I have to tell you that i'm using reactivision since 2012 and i love it! Now, my problem is when I try to adjust the widw 16:9 grid the result is all deformed and it looks like a mess.

This is the grid adjusted:
[image].

And this is the result:
[image]

Thanks for all bro! Have a nice week!

mkalten, 2016-04-19:
This is a know issue, which is already fixed. There will be a 1.5.1 maintenance release soon, and you can already download a test build for mac and windows from here: 

https://github.com/mkalten/reacTIVision/releases/tag/1.5

Please let me know if this works for you, so I can push out the formal release soon.

PabloArty, 2016-04-19:
That's a detail here, I just download the 1.5 master folder and replaced all the files for windows, and it's not working, BUT i don't know if i need to compile the visual studio project to see the updates working.

mkalten, 2016-04-19:
No need to compile, as I said there is also a test build for windows here:
https://github.com/mkalten/reacTIVision/releases/download/1.5/reacTIVision-1.5.1-win32.zip

PabloArty, 2016-04-20:
Well I tested the windows compiled version and all is good now. If I found more bugs I tell you. Thank's bro!

closed 2016-04-20 by mkalten