mkalten/reacTIVision #16: Visual Studio not compiling

jominga, 2016-05-27:
Hello,

I need to run the reacTIVision server on windows, but Visual Studio gives me the following compile error:

> Error C2371: 'int32_t': redefinition; different basic types

The error comes up in TcpSender.cpp in the first code lines:

```
#ifdef  WIN32
#ifndef int32_t
typedef DWORD int32_t;
#endif
#endif
```

Any help?

mkalten, 2016-05-27:
you are brave, I just rewrote the whole windows camera driver implementation.
I also removed this apparently obsolete definition, so everything should work fine now.

BTW, please try the new interactive camera setup feature, by pressing 'K'

jominga, 2016-05-29:
It compiles now, thank you very much!

closed 2016-05-29 by mkalten