mkalten/reacTIVision #21: [linux] [portvideo] does not allow abitary /dev/video* numbers

s-light, 2017-02-15:
if you have a cam that is fixed at a device name with an id higher than 'count' video devices - so the list 
from `$ ll /dev/video*` has spaces in the numbering reacTIVision/portvideo fails to open it as it thinks the devices have to be in order.
i found this in #20.
from a short look there are multiple places in the code that need to be changed.
one point i think is
[V4Linux2Camera.cpp#L261](https://github.com/mkalten/reacTIVision/blob/master/ext/portvideo/linux/V4Linux2Camera.cpp#L261)
` sprintf(v4l2_device,"/dev/video%d",cfg->device);`
if i understand it correct currently in the camera.xml the id attribute in `<camera driver="default" id="0">` means the portvideo internal camera list id. (that makes sens so you can support other non v4l2 cams also..) but it is also used to generate the v4l2 device name.
i currently don't have a good idea how to fix this in an nice way.

[V4Linux2Camera.cpp#L258](https://github.com/mkalten/reacTIVision/blob/master/ext/portvideo/linux/V4Linux2Camera.cpp#L258) is one of the other points where the numbering is directly used that i found..
` if ((dev_count == 0) || (cfg->device<0) || (cfg->device>=dev_count)) return false;`

mkalten, 2017-02-15:
thanks for the heads up Stefan, there seems to be a problem there indeed,
I'll look into it tomorrow I hope ... best, Martin.K

mkalten, 2017-02-16:
@s-light I just pushed a fix for that issue, let me know if this works for you now ...

s-light, 2017-02-17:
that was fast :-)
testested it with an v4l2-loopback device as /dev/video40 and it worked fine :-)
thanks!

closed 2017-02-17 by s-light