mkalten/reacTIVision #23: installation issues on raspbian/(raspi 3)

nylki, 2017-05-17:
Hey,
I am currently trying to install the reactivision_1.5.1-1_armhf.deb on the raspberry pi 3 running the latest raspbian-lite. There are several packages missing, which I'd have to install. However `libjpeg-turbo8` does not exist (anymore?) in raspbian to be installed. Installing, what I think is the replacement package, `libjpeg62-turbo-dev` does not resolve the issue when installing the `.deb`.

Do you have any idea how to fix this?

```.sh
pi@tidreactivision1:~ $ sudo dpkg -i reactivision_1.5.1-1_armhf.deb 
(Reading database ... 43980 files and directories currently installed.)
Preparing to unpack reactivision_1.5.1-1_armhf.deb ...
Unpacking reactivision (1.5.1-1) over (1.5.1-1) ...
dpkg: dependency problems prevent configuration of reactivision:
 reactivision depends on libsdl2-2.0-0 (>= 2.0.4); however:
  Package libsdl2-2.0-0:armhf is not installed.
 reactivision depends on libstdc++6 (>= 5.2); however:
  Version of libstdc++6:armhf on system is 4.9.2-10.
 reactivision depends on libjpeg-turbo8 (>= 1.3); however:
  Package libjpeg-turbo8 is not installed.

dpkg: error processing package reactivision (--install):
 dependency problems - leaving unconfigured
Processing triggers for hicolor-icon-theme (0.13-1) ...
Processing triggers for mime-support (3.58) ...
Errors were encountered while processing:
 reactivision
pi@tidreactivision1:~ $ uname -a
Linux tidreactivision1 4.9.28-v7+ #999 SMP Tue May 16 14:03:19 BST 2017 armv7l GNU/Linux
pi@tidreactivision1:~ $ 

```

nylki, 2017-05-17:
Hm, checking out the git repo and manually building 1.6 does work though (I installed `libturbojpeg1`). I'll close this issue, but it would be nice to have a hassle-free `.deb` for newer debians.

closed 2017-05-17 by nylki

copypastecopypaste, 2017-12-16:
Hey nylki

I am one of those newer debians. And I have no clue how to make the installation of this version of reactivision work on my raspi. Can you give me some instructions please?