mkalten/reacTIVision #24: wrong dimensions from camera

nylki, 2017-05-31:
continuing the other issue: 
`sudo modprobe bcm2835-v4l2` on raspberry pi 3 with latest rasbian and picam2 worked. Tested it with the `motion` tool to capture frames via v4l2.

However when starting reactivision 1.5.1 I get wrong camera (848x32 with a framerate of 30FPS). Setting any manual width/height in the `camera.xml` results in this dimension. When doing `capture width="max" height="max"` it actually works. But the maximum resolution is too much for the raspberrypi to handle.

Is there something I might be missing?

mkalten, 2017-05-31:
you can list all available formats with reacTIVision -l
it's a long list since the raspicam driver only provides stepwise image formats, no standard resolutions ... you should be able to choose any of those combinations shown

nylki, 2017-06-06:
thank you, choosing a resolution form the list and entering the values into the XML worked just fine!

closed 2017-06-06 by nylki