mkalten/reacTIVision #25: linux v4l2 camera setting 'exposure_absolute' not restored correctly

s-light, 2017-06-08:
system:
* kubuntu 16.10 64bit 
* pulled repository  at commit 10744ef
expected behavior:
 1. if camera is set to absolute exposure (done with guvcview -device="/dev/video0" --control)  
 2. this value should be written in the camera.xml config file
 3. on next start the values from camera.xml should be used to set the actual parameters.
current behavior:
 1. + 2. same as expected
3. on restart of reacTIVision camera again start in exposure auto mode.

s-light, 2017-06-08:
update:
with `make debug` the behavior is different for 3 and 4:
3. on next start value is read and set at camera correctly.
4. on exit it is saved correctly

s-light, 2017-06-08:
update2:
with `make debug` and exposure at minimum value the behavior is different for 4 and 5:
4. on exit it is saved as `min`
5. on next start: `min`seems to be not recognized - exposure is switched to auto

mkalten, 2017-06-08:
thanks for testing, I still have been tweaking this particular setting, and although I managed to improve the actual functionality, there are still some apparent problems. I hope to have some time for that next week, your tests are definitely helping to resolve this I guess ...

mkalten, 2017-06-09:
@s-light I just pushed another auto exposure fix which seems to work for me,
let me know if this also resolves your issue ...

s-light, 2017-06-12:
@mkalten i think that partly fixed things - 
now the behavior i see is the same for `make all`and `make debug`
i also found a bug in the Logitech HD Pro Webcam C920 firmware (tested only in combination with v4l2):
if i set absolute exposure with `v4l2-ctl  --device /dev/video19 --set-ctrl exposure_absolute=10` and directly after this read it again:
```bash
$ v4l2-ctl  --device /dev/video19 --get-ctrl exposure_absolute
exposure_absolute: 9
```
so it reports one lower than it is set..

with my other cam this set / get works as expected (it returns the same value as is set)

but the bug from update2 is still there: if i use the min value (in my case its 1) its read correctly and also set correctly. on exit it is saved as `min`. on restart of reacTIVision it seams that this min-value is recognized but for some reasons the cam is set to auto.
if i use some other text like `xxx` the cam is set to the minimum exposure in manual mode.
additionally i see two failed attempts to set uvc control messages in dmesg. - if it successfully sets the cam to absolute exposure i only see one failed attempt.
so i think the bug is because if the `min` setting is handle the cam is not first set to manual exposure mode.

mkalten, 2017-06-12:
ad 1) I guess this may be your camera, which just selects the nearest value?
ad 2) I pushed another fix which hopefully resolves this issue

s-light, 2017-06-13:
2) fixed :-) thanks for your great work!!!
---
1) offtopic: 
tested some more - don't know if this is a bug in v4l2 or the cam firmware itself -
```bash
v4l2-ctl  --device /dev/video19 -l
exposure_absolute (int)    : min=3 max=2047 step=1 default=250 value=3
```
for all values below 500 it returns value one lower than set. i can see a difference in all steps down to 3. (3 also is darker than 4)

closed 2017-06-13 by s-light