mkalten/reacTIVision #26: support multiple reactivision terminals running in the same time in the platform WIN32

KevinQ-BITSSR, 2017-08-24:
https://github.com/mkalten/reacTIVision/blob/5d749515cd7bf5b5fe632f127bf28f3817a649c6/common/FrameThresholder.cpp#L196
The current thread identifier `TEXT("ProcessEvent"+i)` can not distinguish the threads who share the same number `i` in different reactivision terminals, which will leads to a kind of lock-like problem as described in [https://sourceforge.net/p/reactivision/discussion/user/thread/9504717b/](url)

mkalten, 2017-08-24:
could you try to set the event name to NULL, and see if that works?

tdata[i].ghWriteEvent = CreateEvent(NULL, FALSE, FALSE, NULL);

KevinQ-BITSSR, 2017-08-25:
Yeah it worked! Thank you :)

mkalten, 2017-08-25:
OK great, thanks for testing ... I pushed the according fix!

closed 2017-08-25 by mkalten