mkalten/reacTIVision #27: reactivision and pc webcam

simone27zero1, 2017-09-14:
Good evening at all. I need help for my processing project; i'll explain: when i start the sketch in processing the laptop's webcam works but how i open reactivision the laptop's webcam does not work anymore and reactivision tells me it does not find any webcam. i hope you can will help me.

mkalten, 2017-09-14:
try to start reactivision first, so it gets access to the camera.
which processing sketch are you referring to?
you should be using reactivision with the according TUIO client,
which actually does not need any camera access:
https://github.com/mkalten/TUIO11_Processing

simone27zero1, 2017-09-14:
yes i'm using reactivision with the according TUIO client; now i try to start reactivision first

simone27zero1, 2017-09-14:
though i start reactivision first, it tells me the laptop's webcam not found

mkalten, 2017-09-14:
which operating system and reactivision version are you actually using?

simone27zero1, 2017-09-15:
actually i'm using windows 10 pro how operating system and the reactivision version is 1.5.1

simone27zero1, 2017-09-15:
i think it's a problem between the reactivision and the laptop's webcam, beacuse now i start the Tuio demo and reactivision but there is the same problem or that reactivision don't find the laptop's webcam

mkalten, 2017-09-15:
does your camera actually work at all (eg. with skype)?
what is the output of reacTIVision -l (from the console)?

simone27zero1, 2017-09-15:
my camera works at all, now i try to use to make photo and video and it works
where do i view the uotput of reacTIVIsion?

mkalten, 2017-09-15:
open the command console and change to your reactivision directory.
then execute "reacTIVision -l" from the console ...

mkalten, 2017-09-17:
or place this batch file into your reactivision directory and then start it:
https://github.com/mkalten/reacTIVision/blob/1.5.x/windows/list_devices.bat

Judithricketts, 2018-05-03:
Hello, my first question here! I am a Student, new to programming I would like to use reactivision+ Tuio with Processing.org.  I understand I need to use Processing 2.0 (is is correct?)  but cannot get Processing to recognise the presence of the reactivision library. Would you please firstly walk me through the install process then the set up process, laypersons language please....appriciated

closed 2018-05-08 by mkalten