mkalten/reacTIVision #28: SDLInterface not available

JulesTopart, 2017-09-20:
Hi !
I'm stuck on a strange error... I've installed reacTIVision and all the libs on a Raspberrry pi 3,
I'd clone your github repo and then execute `make -d -j6` to compile. Then I got the reacTIVision executable and reacTIVision.desktop (wich doesn't work by the way)
I can run reacTIVision by typing `./reacTIVision` 
then I get that : 
`reacTIVision 1.6 (Sep 20 2017)

camera: UVC Camera (...) (...)
driver: default
codec: yuyv
format: 640x480, 30fps
TUIO/UDP ...
TUIO/TCP ...
TUIO/WEB ...
TUIO/FLC ...
tuio/src rtv@127.0.1.1
SDLInterface not available!

commands ...
`
So everything seems to work as charm but I can't have any GUI ... 
Where this message comes from ? : "SDLInterface not available!"

Thanks for this great project and for your time
Best regards !

JulesTopart, 2017-09-20:
I manage to repair my reacTIVision.desktop But no GUI appears
My camera turn on, reactivision is running and accessible via TUIO but no GUI.
Hope it can help !

mkalten, 2017-09-21:
you'll need to fix your SDL installation first ...
the GUI depends on a working SDL library, which on some Raspi distros is broken.
maybe you just uninstall the SDL packages and compile/install it yourself in /usr/local

closed 2022-10-20 by mkalten