mkalten/reacTIVision #29: feature request: Multi Camera input

s-light, 2017-10-11:
as discussed at https://sourceforge.net/p/reactivision/discussion/user/thread/9504717b/#32be
it would be super to have a basic multi-cam input option.
i will start with the porting of the things mostly already done by cmayer0087 to the simpler approach based on the current master branch. Pull request will come.

This Feature request is meant as discussion platform for 'what should work in which way'.

Current Approach from my side would be:
- [ ] Getting v4l2loopback to work with portvideo to have a stable test-setup ([more info](https://github.com/s-light/webcam_simulation))
- [ ] port/copy MultiCamera Driver 
- [ ] document usage

configuration would be done in camera.xml something like this:
```xml
<portvideo>
    <camera driver="multicam" id="0">
        <capture format="mjpeg" width="600" height="360" fps="30" color="true" flip_h="false" flip_v="false"/>
        <settings/>
        <frame width="600" height="360" xoff="0" yoff="0" mode="-1"/>
        <childcamera driver="default" id="0">
            <capture format="mjpeg" width="320" height="240" fps="30" color="true" flip_h="false" flip_v="false"/>
            <settings/>
            <frame x="100" y="0" width="200" height="200" xoff="-110" yoff="-30" mode="-1"/>
        </childcamera>
        <childcamera driver="default" id="19">
            <capture format="mjpeg" width="320" height="240" fps="30" color="true" flip_h="false" flip_v="false"/>
            <settings exposure="auto"/>
            <frame x="300" y="0" width="300" height="200" xoff="-10" yoff="-30" mode="-1"/>
        </childcamera>
        <childcamera driver="default" id="20">
            <capture format="mjpeg" width="320" height="240" fps="30" color="true" flip_h="false" flip_v="false"/>
            <settings exposure="auto"/>
            <frame x="100" y="200" width="240" height="140" xoff="-20" yoff="-50" mode="-1"/>
        </childcamera>
    </camera>
</portvideo>

```
[image]

the idea is you have a canvas defined by the first  `<capture width="600" height="360" />` tag (green border). 
than for every camera you want to show you create one `<childcamera>` tag. in this element the `<capture>` tag defines the hardware camera config/format. then there is a `<frame x="100" y="200" width="240" height="140" xoff="-20" yoff="-50" mode="-1" />` tag to define the position and visible portion of the camera image.

I have to look into how to integrate this so that it is backward-compatible with the current config.

If you have ideas how to change this for easier understand pleas let me know.

davidsalamon, 2022-10-20:
For anyone else stumbling upon this feature idea: you can achieve the desired output (as outlined by s-light) by using OBS Studio and combining multiple camera sources into a virtual camera output stream. However, OBS Virtual Camera used as input seems to freeze reacTIVIsion (1.5.1)

s-light, 2022-10-21:
> However, OBS Virtual Camera used as input seems to freeze reacTIVIsion (1.5.1)

great idea !!
its a while back - but if i remember correctly i also had freezes with virtual cameras - 
i think it had to do with order of opening and the pixel format that is used...
so maybe it is possible to get it running if you find the correct order of device-opening / accessing and pixel-format options...

Malmaars, 2023-12-17:
@davidsalamon 
> For anyone else stumbling upon this feature idea: you can achieve the desired output (as outlined by s-light) by using OBS Studio and combining multiple camera sources into a virtual camera output stream. However, OBS Virtual Camera used as input seems to freeze reacTIVIsion (1.5.1)

I'm stumbling upon this now. I need to connect my OBS virtual camera to reactivision, but it doesn't seem to be able to recognise it. And I have no clue how to find the correct camera id if it would recognise it (as I need to change the camera id in camera.xml to change camera inputs).
Do you have experience doing this? If so could you help me out?