mkalten/reacTIVision #32: No object in Pure Data    No connect no data

arpeggio1234, 2018-04-26:
hello, i am on Linux, Manjaro, and use ReacTIVision (this works good).
And then i start Pd (Pd-L2ork)and when i load a sketch it says: couldn't create,
so Pd can not create the tuioclient object, 
i used the 1.1.5 zip and the later 1.3 zip and still no succes.
I just load the things from the unpacked zip folder. 
Can someone help me out here please, i searched a lot but still the same.
the tuioclient object will not create.
Do i miss something; a library or is it just a Pd thing, where do i have to put the files.
Many thanks, Dian

mkalten, 2018-05-08:
which platform are you using? maybe you just need to recompile the pd external
1.1.5 is the latest version ...

closed 2018-05-08 by mkalten

arpeggio1234, 2018-05-09:
I got it working. Had to make the tuioclient.pd_linux file 
And then pd could create the tuioclient object.
Great. Thanks for your support.
Dian

https://github.com/mkalten/reacTIVision/issues/32#issuecomment-387331299