mkalten/reacTIVision #34: SUR40 and Windows

sofa95, 2018-06-13:
Hey everyone, i wanted to use reacTIVision on the SUR40 with Windows 7. Everytime i try to start it, i get an error. I know that there is SurfaceToTuio but it doesn't work. Anyone here using reacTIVision on the SUR40 with Windows?

floe, 2018-06-14:
It depends a bit of what you want to achieve: if you want to recognize the Amoeba tags that come with reactivision, then you need access to the raw camera stream, and that only works with Linux. If you just want to convert the touch data to TUIO, then you should ask the Surface2tuio authors first.

mkalten, 2022-10-20:
There is no native SUR40 "camera" driver for reacTIVision on windows yet, this works only on Linux at the moment. On Windows one can use the Sur40Tuio tool, which translates the native Finger and Bytetag tracking to TUIO.

closed 2022-10-20 by mkalten