mkalten/reacTIVision #36: SUR40 and Windows10

PierreMahieux, 2019-03-19:
Hi, I'm currently working with a SUR40 with Windows 10. The objective is to develop a small Unity application where you can control virtual object using real ones with fiducial markers. I thought about using reacTIVision to get the TUIO informations to my application. So I downloaded reacTIVision 1.5.1 and try to get it running but everytime I start it, it quits without any message.
I also tried to make it work using the sources. But when trying to start the compiled program, it crashes with an error message saying (translated from french so maybe innacurate) "Application couldn't start properly (0xc000007b). Presse ok to quit.".
I've seen the issue #34 and tried to use SurfaceToTuio but I got to admit I may have not quite underestand how all of this works.
Has anyone any clues about why it won't work and how to run it ?

floe, 2019-03-21:
Reactivision + SUR40 currently only works under Linux. Surface2Tuio is extremely old and very likely won't work with Windows 10 anymore (was written for Windows 7). So right now, your options are either to try and update Surface2Tuio if you really need to use Windows, or switch to Linux and just use Reactivision.

PierreMahieux, 2019-03-22:
Thank you for your help.
I'll try to use Surface2Tuio on Windows 10 and switch to Linux + reacTIVision if I can't make it work.

closed 2019-03-22 by PierreMahieux

mkalten, 2022-10-21:
Surface2Tuio still works fine on Windows 10 and even 11. You can use it to provide the native SUR40 touch and bytetag detection through TUIO. Using Touchinjector you can even restore native Windows 10/11 multitouch functionality.

https://github.com/floe/surface-2.0/tree/master/tuio/windows
https://github.com/michaelosthege/TouchInjector/releases/tag/1.3