mkalten/reacTIVision #37: mini set not working with reactivision 1.5 ?

blaisegeo, 2019-05-17:
Hi
I installed http://prdownloads.sourceforge.net/reactivision/reactivision_1.5.1-1_amd64.deb on a debian system.
It"s ok with the default fiducials: I can see the green numbers on each default fiducial when put under the cam, but I'ld like to use the mini set of fiducials.
I updated the reacTIVision.xml file with this line:
<fiducial engine="amoeba" tree="mini.trees" />
or this one:
<fiducial amoeba="mini" />
But it seems reactivision can't detect these mini fiducials.
Is there a special trick to use these mini fiducials ?
Would reactivision 1.6 better work than 1.5 ? (Then I'ld have to compile it).
If 1.5 is really not working with the mini fiducials, can it work with the "small" set ?
Thanks in advance

floe, 2019-05-17:
Not sure if I remember 100% correctly, but try adding the full path to the respective .trees file in the XML. I have successfully used both sets with unmodified reactivision 1.5.1.

blaisegeo, 2019-05-17:
Thank you so much, floe, full path to the .trees file made it work !

closed 2022-10-20 by mkalten