mkalten/reacTIVision #42: ReacTIvision not working because of Tangible Interaction Framework

G0NI25, 2022-06-12:
For some weird reason, the program crashes because the Tangible Interaction Framework stops working, I geniunely have no clue what that even is, is it something related to the NET framework?

Also, I wanna point out that I am using Windows 7.

Would gladly appreciate help :
[image]
)

seeacloud, 2022-06-12:
您好，您的邮件我已经收到，我会尽快回复您。谢谢！

floe, 2022-06-13:
@seeacloud would you mind turning off your email autoresponder?

floe, 2022-06-13:
@G0NI25 "Tangible Interaction Framework" is just the long name for the reacTIVision executable. We'd need a bit more information about the crash, but I wouldn't know how to find that on Windows 7 - have you tried starting from the console?

closed 2022-10-20 by mkalten